tpmullan/OctoPrint-DetailedProgress #4: Printer Locking up, LCD Displaying "KILLED" and unresponsive.

SoundDoc, 2020-05-05:
Just tried printing after updating to this branch, Prusa i3 Mk2S.
Previous version worked as expected moments minutes before, as soon as this version was installed and Octoprint restarted the print job would begin, bed and hotend heat, then moving from waiting for bed to heat instead of printing the LCD displayed KILLED and printer became unresponsive to any input. Octopi displayed a communication error and also hung, appeared to restart/reload.
Restarted printer, restarted Octopi, attempted again, same results.
Tried twice more, no difference, disabled the plugin and printing returned to normal (just no detailed progress now).

Relevant part of the Octoprint.log attached up until the start of the print that's currently running without issue if it helps.

[octoprint.log](https://github.com/tpmullan/OctoPrint-DetailedProgress/files/4578688/octoprint.log)

SoundDoc, 2020-05-05:
Found a few related Octoprint posts regarding Checksum, Line number problem, but specifically a post from Foosel regarding this https://community.octoprint.org/t/error-no-checksum-with-line-number-last-line-18/8838/2 

Modified lines 156-171, from:

			etl_format="{hours:02d}:{minutes:02d}:{seconds:02d}",
			print_done_message="Print Done",
			use_M73 = True,
			M73_PrusaStyle = False,
			all_messages=[
				'{filename}',
				'{completion:.2f}% complete',
				'ETL: {printTimeLeft}',
				'ETA: {ETA}',
				'{accuracy} accuracy'
			],
                        messages=[
				'{completion:.2f}% complete',
				'ETL: {printTimeLeft}',
				'ETA: {ETA}',
				'{accuracy} accuracy'

to:

			etl_format="{hours:02d}h{minutes:02d}m{seconds:02d}s",
			print_done_message="Print Done",
			use_M73 = True,
			M73_PrusaStyle = False,
			all_messages=[
				'{filename}',
				'{completion:.2f}% complete',
				'ETL  {printTimeLeft}',
				'ETA  {ETA}',
				'{accuracy} accuracy'
			],
                        messages=[
				'{completion:.2f}% complete',
				'ETL  {printTimeLeft}',
				'ETA  {ETA}',
				'{accuracy} accuracy'

Removing the : characters, re-enabled the plugin and its working properly now.

The linked post is in regards to support for : missing in some firmwares, that "bug" apparently affects the latest Prusa Mk2 firmware available, possibly add a check (I see other Prusa related changes) to modify the string?

tpmullan, 2020-05-05:
I will remove : from the default message.  I see no reason to include it if it is not supported on all firmware's.  I have not had this issue since I am running the latest Marlin.  Thanks for identifying the issue you were having

tpmullan, 2020-05-05:
I created a pull request.  If you could test that and let me know if it works I will release the patch asap

SoundDoc, 2020-05-05:
I'll test it tonight when I get home and let you know (also after I figure out how to apply a pull request).
Just saw the pull with the changes to line 156 etl_format, that was half the issue, the other half was in 163-164, and 169-170, 

				'ETL: {printTimeLeft}',
				'ETA: {ETA}',

also caused the hang, I changed to:

				'ETL {printTimeLeft}',
				'ETA {ETA}',

Thanks for your quick attention.

tpmullan, 2020-05-05:
I updated the pull request to remove those too.

You can install the plugin manually from url in octoprint using:
https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/feature/message_bug.zip

I find that is the easiest way so you don't have to manually pull code and the octoprint ui does all the work.

PR #6 (2020-05-05): Changed defaults for time
This is to resolve #4 and resolve #5 

As shown here: https://community.octoprint.org/t/error-no-checksum-with-line-number-last-line-18/8838/2
closed 2020-05-05 by tpmullan