tpmullan/OctoPrint-DetailedProgress #5: Error: Printer keeps requesting line 6 again and again, communication stuck

dstaas, 2020-05-05:
Running v0.2.3, I get this error when printing from Octoprint:

> There was a communication error while talking to your printer. Please consult the terminal output and octoprint.log for details. Error: Printer keeps requesting line 6 again and again, communication stuck

If I uncheck 'send M73 progress updates' it doesn't help.  

If I uninstall the plugin and restart Octoprint, the same file prints fine.  I'll try going back to an older version.

Thank you for your work on this, please let me know if there's any more information I can provide.

dstaas, 2020-05-05:
Update: I tried the most recent version of https://github.com/dattas/OctoPrint-DetailedProgress and it does seem to be working correctly for me.  (CR-10 with stock firmware)

aireng1, 2020-05-05:
Hi, 
I have the same problem. CR-10 stock firmware. This is the log file fault:
Send: N4 M117 0.02% complete*16
| Recv: ok
| Send: N5 M73 P0*82
| Recv: ok
| Send: N6 M117 ETL: 00:19:01*76
| Recv: Error:No Checksum with line number, Last Line: 5
| Recv: Resend: 6
| Recv: ok
| Send: N6 M117 ETL: 00:19:01*76
| Recv: Error:No Checksum with line number, Last Line: 5
| Recv: Resend: 6
| Recv: ok
| Send: N6 M117 ETL: 00:19:01*76
| Recv: Error:No Checksum with line number, Last Line: 5
| Recv: Resend: 6

agleaveuk, 2020-05-05:
Hi, i'm having the same problem since I updated, as soon as I disable the plugin I am able to print again. Im using a Tronxy P802MA and also tried disabling M73 updates but makes no difference.

tpmullan, 2020-05-05:
I am currently running the latest version of Marlin and not seeing any issues.  I will try and recreate, but I only have my ender 3 to test on.  It sounds like there is an issue with the way it is sending the M117 command.  I wonder why it has an N6 in front.

tpmullan, 2020-05-05:
https://marlinfw.org/docs/gcode/M117.html

As far as I can tell there is not checksum that should be sent with this command directly

tpmullan, 2020-05-05:
I created a pull request.  If you could test that and let me know if it works I will release the patch asap.

I believe the issues is from the ":" in the message, and it is related to issue #4 

To fix this I am changing the defaults, but you can also fix the issue in the plugin settings.

aireng1, 2020-05-05:
Hi tom, sorry but how do I test the new code? Should I have M73 on or off?

tpmullan, 2020-05-05:
You can install the plugin manually from url in octoprint using:
https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/feature/message_bug.zip

This will allow you to run the new code.  The M73 should have no effect here.  That just updates the progress bar if available.  That gcode is not included by default in Marlin and will not be enabled in most firmware.

aireng1, 2020-05-05:
After writing the comment I worked out how to change the code in the plugin. eta_strftime="%-m/%d %-I:%M%p",
			etl_format="{hours:02d}h{minutes:02d}m{seconds:02d}s"
Testing it now. Thanks

mifbody, 2020-05-05:
Tested 0.2.4, still receiving the error, here's the terminal log:

Send: N8 M109 S250*103
Recv: T:248.8 E:0 W:?
Recv: T:248.9 E:0 W:?
Recv: T:249.0 E:0 W:9
Recv: T:249.0 E:0 W:8
Recv: T:248.8 E:0 W:7
Recv: T:248.8 E:0 W:6
Recv: T:248.6 E:0 W:5
Recv: T:247.5 E:0 W:4
Recv: T:247.5 E:0 W:3
Recv: T:247.0 E:0 W:2
Recv: T:246.3 E:0 W:1
Recv: T:246.3 E:0 W:0
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Recv: ok
Send: N9 M117 ETL: 04h22m00s*56
Recv: Error:No Checksum with line number, Last Line: 8
Recv: Resend: 9
Printer keeps requesting line 9 again and again, communication stuck
Changing monitoring state from "Printing" to "Error: Printer keeps requesting line 9 again and again, communication stuck"
Send: M112
Send: N10 M112*16
Send: N11 M104 T0 S0*17
Send: N12 M140 S0*86
Changing monitoring state from "Error: Printer keeps requesting line 9 again and again, communication stuck" to "Offline (Error: Printer keeps requesting line 9 again and again, communication stuck)"
Connection closed, closing down monitor
Closing down send loop

tpmullan, 2020-05-05:
So it looks like I need to remove the colon everywhere and it should work.  I will update the pull request again.  Thanks for the help testing

tpmullan, 2020-05-05:
I have a new commit removing all the colons.   If anyone can test, you can install the plugin manually from url in octoprint using:
https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/feature/message_bug.zip

aireng1, 2020-05-05:
Seems to be working now thanks. Could you make an option to have the ETA format as day, month , time please instead of month, day, time. Also could you make an option to output the time in 24 hour?
Thanks

tpmullan, 2020-05-05:
That is already an option in the settings. It uses pythons date format. It probably could be documented better, but that is from the original plugin

tpmullan, 2020-05-05:
@aireng1 I updated the [README](https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/feature/message_bug/README.md) in the pull request to include info on strf time.  Let me know if that helps explain the settings better.

aireng1, 2020-05-05:
That's great Tom. All sorted now, changed my ETA format to %-H:%M %d/%-m. I had to put a decimal at the end or the last digit gets cut off. It happened on old version too but I never bothered doing anything about it. Any ideas why it does it, it does it on ETL,ETA and percent.

tpmullan, 2020-05-05:
I have no idea why it would do that, but my guess would be an issue with the way the firmware is processing the M117 command.  I would compile Marlin manually for your machine and try it on the latest version.  That fixed all my issues on my Ender 3, but that is pure speculation.

closed 2020-05-05 by tpmullan

aireng1, 2020-05-05:
I'd say you're right but I know if I go messing with Marlin I'll break something. I have it all working now. It doesn't print the dot after the text. Just need to enter one after the "complete" so that the e isn't cut off. Any chance in future update you could do that?
thanks

tpmullan, 2020-05-06:
That is also already there. Just update the Print done message in the settings.

aireng1, 2020-05-06:
Sorry the complete, as in % complete not the print done message

tpmullan, 2020-05-06:
That is also in the settings.  Just add a custom message the same as displayed for the current complete with the period at the end.  Then select that instead of the original message.

[image]

aireng1, 2020-05-06:
Gotcha, A lot easier. Thanks. New to this coding craic.

PR #6 (2020-05-05): Changed defaults for time
This is to resolve #4 and resolve #5 

As shown here: https://community.octoprint.org/t/error-no-checksum-with-line-number-last-line-18/8838/2