tpmullan/OctoPrint-DetailedProgress #8: only sending ETA

Mac2712, 2020-05-08:
HI I have just loaded the latest version of DetailedProgress looks promising. 
Its only adding the ETA line into the Gcode stream to the printer running Merlin 2.0.1, old version worked fine before

jhillyerd, 2020-05-09:
I have a similar problem, mine is configured (according to web UI) to send:

```
{filename}
ETL {printTimeLeft}
ETA {ETA]
```

but all that shows on my printers screen is `Good accuracy`

This is what I see in config.yaml:

```
    detailedprogress:
        all_messages:
        - '{filename}'
        - ETL {printTimeLeft}
        - ETA {ETA}
        messages:
        - '{accuracy} accuracy'
        time_to_change: '5'
```

So probably related to #7 as well.

tpmullan, 2020-05-14:
I will look into this.  I have only set the config from the settings.  It sounds like it is not updating the config file

tpmullan, 2020-05-14:
I do not see anywhere in the code that is directly interfacing with the config file.   It looks like it is imported as a dictionary and the defaults are added.  I personally do not ever manually edit the configuration file since it is a pain, and only use the web settings, so I have not seen this bug.  Currently I am considering completely deprecating editing the config file.

Can someone that is facing this issue test removing those sections from the configuration file then using the web interface only to make changes?

jhillyerd, 2020-05-14:
I just tried deleting ETL via the web interface, and it updated to:

```
   detailedprogress:
        all_messages:
        - '{filename}'
        - ETA {ETA}
        messages:
        - ETL {printTimeLeft}
        time_to_change: '5'
```

Prior to https://github.com/tpmullan/OctoPrint-DetailedProgress/issues/8#issuecomment-626182399 - I had deleted the accuracy line via web, so for some reason it overwrites messages: with the deleted entry.  The content of all_messages is correct (in terms of what I want the be displayed), but is not used.

I will try removing the detailedprogress section from the config entirely.

tpmullan, 2020-05-14:
So I verified on my end that if I remove / add an item in the config it shows from the web interface.  And if I change in the web interface and click save on the settings screen, it shows in the config.  I can not reproduce this issue.

What version of octoprint is being used?  Obviously many people are seeing this issue.

tpmullan, 2020-05-14:
One thing to note is the web settings show all messages and the selected ones are active.  You must control + click all messages you want:

[image]

jhillyerd, 2020-05-14:
Oh.  I think that is likely what is confusing everybody then.  Labeling the <select> box "Active messages" implies that all of them are active.  I'd recommend labeling it "Available messages" and then add some text stating that the selected ones are the active messages.

Perhaps even a warning if the user tries to save with no messages selected, as that's what I've been doing.

tpmullan, 2020-05-14:
I can add a note in the README and change the name of that field in the settings.  Since this is code that I have merged from others I have not personally touched it and just kept what they had since it worked for me.  I agree though that I think this is just a user experience bug and with a more intuitive settings page and better docs we should be good to go.  If everyone facing this issue can respond here I can make updates based on feedback.

Mac2712, 2020-05-14:
I think that a clearer message needs to be on the web page as not everyone reads the readme

Mac2712, 2020-05-14:
OK just reloaded this version and with the correct setting it now is working Just need to re download marlin for M73. I would strongly suggest you get something on the web page to clarify how to configure this and get that uploaded so Octoprint updates all those users how haven't **fully understood.**

Mac2712, 2020-05-14:
Update I have noticed that its sending the progress at a 10th of what the webpage shows
11% on webpage 1.1% on the Gcode stream and display

tpmullan, 2020-05-14:
That is interesting.  From what I can see the percentage code has not changed from the original version.  My printer again is showing the same value as the web site.  It almost sounds like there is a broken formatter.  @Mac2712 can you share a screenshot of your config or paste a copy here?

Mac2712, 2020-05-14:
[image]

tpmullan, 2020-05-14:
Thanks, that looks good from the config side.  Let me know if it continues to have the wrong value.  On really fast prints it can be a few seconds slow, but it should be really close to the value displayed on the site.  I will try to reproduce, but if you can open another bug we can track that issue on another thread.  I would like to keep the displaying messages issue on this thread.

tpmullan, 2020-05-19:
I added some code, but still need to test it a little more.  If I could get some help testing I can release it faster.  You can install it manually from the install plugin page with the link https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/feature/update-0.2.5.zip

tpmullan, 2020-05-19:
Updated settings current look like this:
[image]

closed 2020-05-22 by tpmullan