tpmullan/OctoPrint-DetailedProgress #10: Messages are not correctly being managed

PRGeno, 2020-05-10:
Since this fork became available, I've had all kinds of issues with the display on my two Prusa Mk3s Bears running 3.8.1 firmware.

Through investigation, it appears the plugin section of the config.yaml file is not being correctly updated.  There are two message sections appearing in the config file: "all_messages:" and "messages:" 
I'm not sure of the purpose of the two message sections but neither of them seems to get updated correctly when making any changes in the plugin configuration within the Octoprint UI.

I made several changes to the plugin from within the Octoprint UI and then checked to see what the config.yaml file contained.  I was never able to see where either of the two message sections matched what I configured in the Octoprint UI.

I'm now playing with manually edited the config.yaml file to see if I can figure out what it is supposed to happen with various options.  But no conclusions yet.  The bottom line is this plugin no longer works as expected.

PRGeno, 2020-05-10:
It seems the "messages" section" updates with the last message that was deleted, and that is all that is then displayed on the printer.  I may take a look at the code when I get some time, this would seem to be a bug that should be pretty easy for the author to track down and quickly fix.

PRGeno, 2020-05-11:
I've tried all kinds of testing of various combinations of settings in the UI and config.yaml editing.

Nothing has worked and this release appears to be broken at a code level beyond where any configuration change can overcome.

I finally reverted back to the original dattas version this was forked from and all is well again.  Now if I can only get rid of the announcement telling me to switch over to this broken fork.

Mac2712, 2020-05-13:
Hi PRGeno
Where did you get the original and how did you install I have only used the Octoprint Plugin Manager  
I removed it and now cant go back

PRGeno, 2020-05-13:
@Mac2712 Go to Dattas's original github project from which this was forked from (or at least a failed attempt to fork from) as listed right below the name of this project.  Copy the link to the master.zip and paste it into the Plugin Manager ....From URL field and hit install.  To save you time, here is the URL to the old plugin:

https://github.com/dattas/OctoPrint-DetailedProgress/archive/master.zip

Copy and paste that into the Plugin manager after fully uninstalling and removing the data for this broken one.  Then you will have to manually add the messages you want back into the config.yaml file and you should be good to go.  Just be aware the old plugin doesn't support Python 3.x, but that isn't a real issue as of yet.

tpmullan, 2020-05-14:
I am going to close this and track it under #8 since it looks like the same issue

closed 2020-05-14 by tpmullan

tpmullan, 2020-05-14:
@PRGeno this is not my job, and you can wait a couple days for a response.  4 days during a work week is not a reason to assume a project is abandoned.  Will you please use issue #8 for further comments and test removing everything from the config file.  I will try and reproduce this issue by adding a config file on my printer, but see as this is being developed for FREE, it will probably be this weekend.