tpmullan/OctoPrint-DetailedProgress #11: Progress showing 1 /10th of actual

Mac2712, 2020-05-14:
My display is showing 0.95% complete when its showing 9.5 on the web page

from the Gcode stream 
Send: N1010 M117 0.99% complete*38
Recv: ok
Send: N1011 M73 P0*102

Mac2712, 2020-05-14:
Don't think this is a format problem as the web page is showing 72% and m117 is 55%
its a 32bit motherboard so there should be no overloading of the comms

tpmullan, 2020-05-19:
I also have a 32 bit motherboard and have tested on a 16 bit board. I am unable to reproduce.  What is the total time of the print you are performing?

tpmullan, 2021-01-19:
Closing for inactivity.  I was not able to reproduce the issue on my system.  If you run into this again / have more info feel free to reopen this issue.  As always pull requests are welcome.

closed 2021-01-19 by tpmullan

bthaler1, 2021-02-05:
I'm having this issue as well. I just installed the plugin and this is my first print using it. LCD showing 2.35%, Octoprint web interface showing 12%. This print is supposed to take about 12h to complete. Running on Octoprint 1.5.3 and printer is an Ender 3 Pro running Marlin bugfix-2.0.x (Feb 3 2021). Please advise if I need to do anything on my end thanks.

bthaler1, 2021-02-06:
Update... Strangely, some time after my print got within 2h of finishing, the percentage is much closer. 83.17% and ETL 1h52m50s on the LCD compared to 84% and 1:51:23 in Octoprint. It has shown "Good accuracy" for the entire print. I'm not sure exactly when it started displaying the more correct percentage, but it was probably close to 2h ETL. Hopefully these clues point you in the right direction.