tpmullan/OctoPrint-DetailedProgress #12: M73 progress bar doesn't go to 100%

a1b0hph0b1a, 2020-05-15:
When the M73 messages are enabled, the progress bar doesn't go all the way to 100%, it stops at about 95% when the print is done.

tpmullan, 2020-05-15:
I don't think the code it is using to send status updates is triggered at 100%  I can look into modifying the code, but this will be low priority.  Feel free to submit a pull request.

a1b0hph0b1a, 2020-05-16:
No rush. Just thought I'd mention in case you didn't know. I'm in no rush, can use the old M73 plugin in mean time.

tpmullan, 2020-05-19:
I added some code, but still need to test it a little more.  You can install it manually from the install plugin page with the link https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/feature/update-0.2.5.zip

PR #13 (2020-05-22): [WIP] Feature/update 0.2.5
Updating settings with documentation, and starting progress bar completion update.

Fixes #12 and #8, #7, #9, #10
closed 2020-05-22 by tpmullan