tpmullan/OctoPrint-DetailedProgress #14: Layer number

vockleya, 2020-05-21:
This plugin is just about perfect, but there's just one thing that I feel is missing. Before I found this plugin, I used Cura to add GCode M117 lines with the current layer and total layers, but now that is so longer an option, as this plugin overwrites that.

I'd like to suggest this as a feature for a future version. I'm not sure of the feasibility of this, as a quick look at the Octoprint API doesn't show any way to retrieve this information, but it must exist somewhere, as the GCode Viewer displays this information.

tpmullan, 2020-05-21:
I would suggest using something like: https://plugins.octoprint.org/plugins/PrintTimeGenius

Using the layer the printer is currently on for time estimation is not very useful since density can vary greatly in prints.  I am not sure what value this would add, but if I have some time I can look into it.  Feel free to submit a pull request if you figure out how to add that feature.  I see no harm in adding it, but also see very little gain for me personally.

OllisGit, 2021-01-13:
Hi @vockleya ,

did you try: 
- https://plugins.octoprint.org/plugins/layerdisplay/
- https://plugins.octoprint.org/plugins/DisplayLayerProgress/

The last one provides an api for reading the layer-informations.

tpmullan, 2021-01-25:
Closed by merging #32

closed 2021-01-25 by tpmullan