tpmullan/OctoPrint-DetailedProgress #17: Incompatible with Python 3.8.3

smorar, 2020-06-13:
I'm running octoprint in a relatively unconventional environment - arch, running Python 3.8.3.
When I install this plugin, I get the following error in the plugin manager:
"Incompatible with octoprint or the underlying python environment"

Any advice on further debugging this issue to assist in a fix that will lead to forward compatibility would be appreciated.

tpmullan, 2020-06-20:
So I have it set to less than python 4 here:
https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/master/octoprint_detailedprogress/__init__.py#L202

So it should at least try and run the plugin instead of getting an incompatibility issue.  Have you tried manually installing from the repo?  If not go to plugins and install from URL.  Then give it this url: https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/master.zip

tpmullan, 2020-10-13:
Is this a duplicate of #19 ?  If so it might work now.

tpmullan, 2020-10-22:
Closed for now since it looks like the code change should address this issue

closed 2020-10-22 by tpmullan