tpmullan/OctoPrint-DetailedProgress #19: Won't run under Python 3 due to indentation problems

ManuelMcLure, 2020-07-16:
I upgraded my OctoPrint setup to use Python 3 (3.7.3) and the Detailed Progress plugin fails to load with the following message in octoprint.log:
```
2020-07-16 14:24:46,433 - octoprint.plugin.core - ERROR - Invalid syntax in /home/pi/oprint/lib/python3.7/site-packages/octoprint_detailedprogress/__init__.py for plugin detailedprogress
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 599, in _parse_metadata
    root = ast.parse(f.read(), filename=path)
  File "/usr/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_detailedprogress/__init__.py", line 45
    currentData = { "progress": { "completion": 100, "printTimeLeft": 0 } }
                                                                          ^
TabError: inconsistent use of tabs and spaces in indentation
```

edilsoncorrea, 2020-09-14:
I managed to solve this issue by dowloanding the zip file and opening __init.py with Pycharm Community Edition. There you can fix identation problem. One can then create a new zip file and install it on Octoprint .

Dazag, 2020-09-17:
Hi everyone! @edilsoncorrea can't you make a pull request here with those changes?

ManuelMcLure, 2020-09-17:
There's already a pull request (#20) for this.

jneilliii, 2020-09-20:
Yeah, I opened the PR on 8/19... @tpmullan just needs to merge and bump version.

Dazag, 2020-10-12:
Hi @jneilliii, isn't it possible to install your changes directly from your repo, since this is taking way too long to be merged?

jneilliii, 2020-10-12:
It is.

tpmullan, 2020-10-13:
I merged the changes to master and will bump the version and release tomorrow after some testing.  For now you can install directly from master if you want to test

PR #20 (2020-10-13): Fix identation errors
resolves #19
commit 70242a94 (2020-10-13): quick fix for indentation errors, resolves #19
closed 2020-10-13 by tpmullan