tpmullan/OctoPrint-DetailedProgress #21: When filament runs out pause updates!

Eebel, 2020-10-16:
If a print gets paused due to a filament run out, this plugin still puts the progress information into the printer's buffer.  After the filament is changed and the print resumes I've had the hot end sitting in one place for ten minutes while the screen is updated with eight hours of progress information while the printer was idle sitting idle (the filament ran out right after I went to bed).  This ruined the print.  Please stop processing the remaining time when the print is paused so it does not create this unwanted behavior. Or, keep sending it to the screen so it doesn't create five megabytes of info that has to be sent.

Thanks,

Tim

bennymamo, 2020-10-20:
This happens also when doing an M600 for filament colour change, which is basically the same command I think when filament runs out. I've had huge blobs and am having to diable the plugin when I know I'm doing a colour change.

tpmullan, 2020-10-22:
So when that command is issued I guess the display stops receiving the estimates?  I do not use this command since I do not have that hardware.  Feel free to submit a pull request.  I will look into how to respond to a pause so it does not queue messages, but this is currently completely out of the design of this plugin.

bennymamo, 2020-10-22:
I am not very experienced with python, github, octoprint plugins, etc. I will try my hand at something and create a pull request If someone can check it over.

tpmullan, 2020-10-23:
I would be happy to review any code and help where I can. I just don't have the hardware to test.

bennymamo, 2020-10-23:
I have opened this pull request, not sure if done correctly. https://github.com/dattas/OctoPrint-DetailedProgress/pull/48

Eebel, 2020-10-23:
This problem happens on every printer I have (all five of them).  I don't think it is hardware specific.

bennymamo, 2020-10-24:
Just realised I opened on wrong fork. now opened on https://github.com/tpmullan/OctoPrint-DetailedProgress/pull/23

closed 2020-11-05 by tpmullan