tpmullan/OctoPrint-DetailedProgress #22: ETA not updating with new parameters

msteele999, 2020-10-17:
OctoPrint 1.4.2
Python 3.6.9

When changing settings in the eta_strftime field, they are not reflected on the LCD display. LCD is also reporting UTC time even though system time is set to America/New_York.

[image]

tpmullan, 2020-10-22:
So it looks like you have a 24 hour clock selected.  Please save the settings with ETA selected in the multiselect then take a picture of the lcd so I can see what is displayed.  I do not have enough information to investigate this issue.  I am not seeing this result on my printer.

dryphi, 2021-01-12:
How do you change the clock? What's the code?
I'm getting the wrong ETA time on mine although the ETL is correct.
Is there a way to change the time zone?

tpmullan, 2021-01-13:
It should be going off the system clock. Look up setting your time on Linux.

dryphi, 2021-01-27:
I'm only ever using OctoPi. Shouldn't it auto-detect?

tpmullan, 2021-01-27:
It probably uses ntp to sync the time but has no idea what timezone you have.  That would need geolocation and at best is a guess.

Please try:
https://howchoo.com/pi/how-to-set-the-timezone-on-your-raspberry-pi

tpmullan, 2026-05-25:
I opened #54 with a partial fix for this issue: the plugin now refreshes cached settings during progress updates, so changes to message and ETA format settings should be picked up during an active print. The timezone part still depends on the timezone seen by the OctoPrint process itself, so that portion is not fully resolved by the PR.