tpmullan/OctoPrint-DetailedProgress #24: IP adress is showing extrenal not internal on one machine

robocog, 2020-10-30:
Hi- I have 3 machines - all run this great plug-in
However I decided to "backup using dd" on one machine so I have an image I can use should a card or install go bad
I flashed a card with this image backup and thought it wise to check it boots and runs etc
I did this - and everything seems fine
I put the original sdcard back in to the pi3, and booted up
I noticed the IP address it is showing me on screen now is the external IP of the router - where it used to show the internal LAN IP
The other 2 installs are showing the internal IP
Cannot see anything fishy - and the machine can still be ssh'd and the web interface is still on its same internal IP (router assigns a static IP based on the MAC address for all 3 machines)
Have rebooted the problem one several times, I even swapped octoprint sd cards back to the backed up / restored one - and that is also showing the external IP address

I may delete the mac address association in the router (though it must be working as it is getting the same fixed IP each reboot)
short of that I am out of places to look to try and get it showing the internal IP address again

To make the backup I had to sudo apt install ntfs-3g, mount the usb drive, dd'd it and then installed/used pishrink.sh to make it a reasonable size, unmounted the usb drive...none of that /should/ have affected the IP address though?, but this is exactly when I had the change in what IP it shows :/
Not a major deal - but an annoyance, as it was all working fine till I tried making a backup

tpmullan, 2020-11-05:
I will have to look at the code, but I think this plugin should not affect which ip shows up.  I think it gets that info from somewhere and does not calculate anything.  If you find out what is going on and there is something wrong with the code here we can fix it.  Keep me updated.

hapklaar, 2020-12-06:
How can I prevent it from showing the IP address? My config.yaml only contains the following but it still displays my IP upon powering on the printer

    detailedprogress:
        eta_strftime: '%-d-%m %-H:%M%p'
        etl_format: '{hours:02d}:{minutes:02d}:{seconds:02d}s'
        use_M73: false

Lecl0ne, 2020-12-23:
> How can I prevent it from showing the IP address? My config.yaml only contains the following but it still displays my IP upon powering on the printer
> 
> ```
> detailedprogress:
>     eta_strftime: '%-d-%m %-H:%M%p'
>     etl_format: '{hours:02d}:{minutes:02d}:{seconds:02d}s'
>     use_M73: false
> ```

Very interested about it too. I want to disable IP address display. Any chance to get a tickbox for it ? :)

tpmullan, 2021-01-19:
I think that message is hard coded to be displayed when the printer boots.  If you want to make that programmable it should be possible, but will need a code change.  I would be happy to accept a pull request.

Lecl0ne, 2021-01-20:
I didn't have that message displaying before installing your plugin sir.

tpmullan, 2021-01-25:
Found the line of code that displays the IP address on the screen:

https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/master/octoprint_detailedprogress/__init__.py#L51
```python
			self._printer.commands("M117 IP {}".format(ip))
```

If we add a variable to make that optional then we could have it not print the IP.  

It is currently just asking the system for the ip using a convoluted python library call:

https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/cb085042144bc5532d208e1f030fa98fe725e284/octoprint_detailedprogress/__init__.py#L157

I will look into adding a tick box, but you are more than welcome to make the change and submit a pull request.

Lecl0ne, 2021-01-26:
I'm afraid I have no idea how to do it...

tpmullan, 2021-01-31:
I added the code to the release that I just made that will make displaying the ip on start optional.  If that is good enough to close this bug please let me know and I will close the issue

Lecl0ne, 2021-02-01:
Hey,
Now it sends "Octoprint Connected".
Can you make the message customizable?
And is there a way it doesn't print anything?