tpmullan/OctoPrint-DetailedProgress #25: 404 Error updating

nickm324, 2020-12-01:
When updating from within Octoprint it fails

Now updating DetailedProgress Plugin to 0.2.6
+++++++++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip --no-cache-dir
Collecting https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip
HTTP error 404 while getting https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip
Could not install requirement https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/dattas/OctoPrint-DetailedProgress/zip/0.2.6
Could not install requirement https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/dattas/OctoPrint-DetailedProgress/zip/0.2.6 for URL https://github.com/dattas/OctoPrint-DetailedProgress/archive/0.2.6.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

Yvrogne, 2020-12-05:
Reinstall it :

Notice regarding plugin "Detailed Progress Plugin"
Affected versions: 0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4
Version 0.2.3 of this plugin is available from a new maintainer, but your version still looks for updates at the repository of the old maintainer. Please re-install the plugin from the repository to update.

tpmullan, 2021-01-19:
Yeah it needs to be reinstalled since this is not the original git repo.  I merged many improvements from many devs to this repo

closed 2021-01-19 by tpmullan