tpmullan/OctoPrint-DetailedProgress #26: Progress sent when printer is changing filament

coliss86, 2020-12-11:
Hi, 

thanks for your work !

I have a strange behaviour : after resuming a filament change with M600, the printer is "paused" : I see a lot of update on the screen of the ETA. During this, the nozzle is touching the print and is melting it. After then, it continues printing as usual. 
I think is displaying all M117 instructions sent during the filament change.

It would be great if this plugin will be able to detect that the printer is paused in order to stop sending M117 instructions.

Cheers

bennymamo, 2020-12-13:
I noticed the same behaviour, and this could be fixed with https://github.com/tpmullan/OctoPrint-DetailedProgress/pull/23

I'm still waiting for @tpmullan to complete tests and if successful push out a minor release.

coliss86, 2021-01-19:
@bennymamo do you have any update ?

bennymamo, 2021-01-19:
Haven't heard from @tpmullan for ages. I cannot release anything myself sorry.

tpmullan, 2021-01-19:
I have not had time to test the pull request that you made, but I did merge it to master.  @coliss86 can you confirm it works for you?  If you do I can just bump the version and create a release off master

tpmullan, 2021-01-19:
To install manually you can install the plugin from url using: https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/master.zip

tpmullan, 2021-01-25:
@bennymamo is this feature working from master?  I am planning a release for this afternoon and will close this bug after.

bennymamo, 2021-01-25:
@tpmullan I'm afraid the changes did not work correctly on my machine. Looked for some more info it seems like M600 does not produce a pause, but a "wait"? It will work if M600 is added as a pausing command in firmware options in octoprint, but that would need you to enter the GUI and resume from there every time.

In short, no, it did not work correctly as expected.

closed 2024-06-16 by coliss86