tpmullan/OctoPrint-DetailedProgress #27: can not handle delta-printers ?

mameniac, 2020-12-17:
I'm using this on my Ender3 and its working fine, but on my Delta printer it always say "Bad accuracy" and showing print times wich are way off that, what the printer really need. Printer is running latest smoothieware, that maybe cause the problem ?

Bougakov, 2021-01-18:
I have a similar issue. The root cause is that the printing speed for layer 1 is much slower than the rest, so initially, the forecast is too pessimistic (days), then it gets back on track.

So it would be a good idea to discard the 1st layer from the calculations.

tpmullan, 2021-01-19:
This plugin just displays the estimate and does not have anything to do with generating the value.  I had the same issue with my Ender 3 pro. I would suggest using https://plugins.octoprint.org/plugins/PrintTimeGenius/ to get better estimates.

closed 2021-01-19 by tpmullan