tpmullan/OctoPrint-DetailedProgress #28: Plugin personalisation does not work

magellan-13016, 2020-12-22:
Hello,

I'm trying to put only one line message : 
{completion:.2f}% Reste {printTimeLeft}

In English, this mean 
{completion:.2f}% Remain {printTimeLeft}

After saving change, some lines are duplicated, other are missing...

Sop you tell us there is a way to manually make changes in a file but where is it and what's his name please ?

Thanks for help.

David.

magellan-13016, 2020-12-22:
For people who want the same thing, here is the way :
1) Download the plugin here : https://github.com/tpmullan/OctoPrint-DetailedProgress/archive/master.zip
2) Extract the file" __init__.py" located in the folder "octoprint_detailedprogress".
3) Modify the lines below : 

	```
##~~ Settings
	def get_settings_defaults(self):
		return dict(
			time_to_change="1",
			eta_strftime="%-m/%d %-I.%M%p",
			etl_format="{hours:02d}:{minutes:02d}:{seconds:02d}",
			print_done_message="Print done",
			use_M73=True,
			M73_PrusaStyle=False,
			all_messages=[
				'{filename}',
				'{completion:.2f}% {printTimeLeft} left',
				'{completion:.2f}% {printTimeLeft} left',
				'{completion:.2f}% {printTimeLeft} left',
				''{completion:.2f}% {printTimeLeft} left''
			],
			messages=[
				'{completion:.2f}% {printTimeLeft} left',
				'{completion:.2f}% {printTimeLeft} left',
				'{completion:.2f}% {printTimeLeft} left',
				'{completion:.2f}% {printTimeLeft} left'
			]
		)
```
4) Put the edited file back in the archive
5) Install the plugin manually by selecting the archive in the plugin manager.

Enjoy !

P.S : It would be great if the plugin setup module could work to avoid all this things ! ;-)

tpmullan, 2021-01-19:
So the interface is a little odd, but you need to add the messages you want with the add message field.  Then you select all the ones you want displayed with the multi select and save.  I did not create the we interface and it uses the built in constructs.  After reading the code it seemed simple to me, but I have heard from many people that it was not intuitive.

tpmullan, 2021-01-19:
Please read over #8 and let me know if that helps

closed 2021-01-19 by tpmullan