tpmullan/OctoPrint-DetailedProgress #29: Custom messages got duplicated after saving

Lecl0ne, 2020-12-23:
When I add some custom lines and save, it duplicates... If I want to delete the duplicated one, it deletes the other too.
I just can't add this custom lines.

I create the lines:

[image]

Then save. If I go back:

[image]

tpmullan, 2021-01-19:
Could you provide more information on your setup?  I am not seeing these issues, but am running on a pi with python 2.  I need to update to python 3, but everyone has a different setup.  In general I would recommend keeping everything as up to date as you can.

Lecl0ne, 2021-01-20:
OctoPrint 1.5.3
Python 3.7.3 (just updated)
OctoPi 0.17.0

I've done a fresh installation.
Recreated my custom lines:

Réalisé à {completion:.0f}%
Reste {printTimeLeft}
Fin à {ETA}

Saved
Restarted Octoprint

Verdict: Lines vanished again.

tpmullan, 2021-01-25:
The lines have to be highlighted when you save to be shown when printing, but even if they are not then they should not vanish.  I will take some time in this month to update to python3.  That might have an effect even though it should not.  The odd part here is the plugin is only using core features to save information.  Nothing was actually created in this plugin.  Are you having any issues saving information in other plugins?

Lecl0ne, 2021-01-26:
I have no problem for saving in other plugins. It's not Python related as the problem was the same with Python 2 few weeks ago.

tpmullan, 2021-01-31:
I just updated to python3 since I have been meaning to for a while.  I am unable to create duplicates on both python2 and python3.  There has to be something that is different here.

I also have:
OctoPrint 1.5.3 
Python 3.7.3 
OctoPi 0.17.0

Lecl0ne, 2021-04-06:
Problem is still exactly the same untill now.
Everything is duplicated twice and I can't delete one line without deleting the same other one.

Alokin-ITA, 2021-09-24:
I'm experiencing part of this bug too, my custom lines have disappeared: I simply added the lines as suggested in your readme, Layer {layerProgress} and Height {heightProgress}
However something weird might have caused this: when I installed DisplayLayerProgress, I wrongly added a new line to your plugin,
[progress]% L=[current_layer]/[total_layers]

I them realized and deleted it, then added the above 2 lines and saved. I'm now printing and the printer shows only 2 messages, printing object 0 and the literal [progress]% L=[current_layer]/[total_layers].

Detailed progress has standard config.
OctoPrint 1.6.1 Python 3.7.3 OctoPi 0.17.0
On python 3 since install.

Lecl0ne, 2021-09-30:
Still buggy for me too.. I disabled it.