tpmullan/OctoPrint-DetailedProgress #31: [Feature request] Show current layer/layer count

jonsag, 2021-01-16:
Hi

For me a good addition to the output on the display is also show current layer/layer count,
ie something like 'Layer 123/456'.

Or perhaps this is already possible and I've missed it?

Anyway, thanks for this great plugin
/jon

tpmullan, 2021-01-19:
Currently we have
```
    # Messages to display. Placeholders:
    # - completion : The % completed
    # - printTime : A string in the format "HH:MM:SS" with how long the print is going
    # - printTimeLeft : A string in the format "HH:MM:SS" with how long the print still has left
    # - ETA : The date and time formatted in "%H:%M:%S Day %d" that the print is estimated to be completed
    # - filepos: The current position in the file currently being printed
    # - filename: The filename of file currently being printed
    # - accuracy: Accuracy of the time estimates
```

So this would be a new feature.  I currently do not have time to implement this, but would happily accept a pull request.

jonsag, 2021-01-19:
No problems.

I'll look into it.

/jon

tpmullan, 2021-01-19:
It also looks like this might be a duplicate of #14 and there are some examples of projects that use the variable that you want.

And this is where you would add your layer variable to the message.
https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/master/octoprint_detailedprogress/__init__.py#L136
```python
		return message.format(
			completion=currentData["progress"]["completion"],
			printTime=currentData["progress"]["printTimeString"],
			printTimeLeft=currentData["progress"]["printTimeLeftString"],
			ETA=currentData["progress"]["ETA"],
			filepos=currentData["progress"]["filepos"],
			accuracy=currentData["progress"]["accuracy"],
			filename=currentData["progress"]["filename"]
		)
```

We are currently getting the state data from https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/master/octoprint_detailedprogress/__init__.py#L68 which is provided by https://docs.octoprint.org/en/master/modules/printer.html#

Good luck 👍

jonsag, 2021-01-21:
Added a pull request, #32

jonsag, 2021-01-23:
I've also added some code to #32  , since there was more info available in DisplayProgess events, in. Sorry for messing a bit with some unnecessary commits...
/jon

tpmullan, 2021-01-25:
Close by merging #32

closed 2021-01-25 by tpmullan