tpmullan/OctoPrint-DetailedProgress #36: M73 "R" is also supported on Marlin

Sebazzz, 2021-03-03:
I was checking if this plugin sends `M73 R` commands for remaining time, and this supported in Marlin as well. If Marlin was compiled not to support it, the R argument is ignored.

https://github.com/tpmullan/OctoPrint-DetailedProgress/blob/467a3a7a2dfcdc02e5074e5066b48d1e8eefeb71/octoprint_detailedprogress/__init__.py#L86-L92

I noticed the flag `_PrusaStyle` but there doesn't appear to be an user interface for it.

I see two solutions:

- Make `_PrusaStyle` available for configuration. This is a simple template change I think.
- Add the R argument to the M73 command

Please let me know if I can help.

tpmullan, 2021-03-04:
That should be a fairly simple change.  If you want to submit a PR just to set that variable from the UI that is fine.  Basically you could just add a variable that is checked in the same if statement that is defaulted in the same way the existing value is used.  You might even be able to expose the existing variable.  Without looking into it I am not sure which way would be better.  Otherwise I will look into implementing this in the future, but don't have a lot of time with everything going on at work this month.

Sebazzz, 2021-03-05:
I will submit a PR. After some further investigation, I will build a separate option for it, because the Prusa option emits M73 twice for some reason, and it also seems more user friendly to allow the user to explicitly state that they want an R parameter emitted.

tpmullan, 2021-03-05:
I am also fine with deprecating the Prusa option in favor of calling out the Added R parameter and making it a little more user friendly.  I have an Ender printer, so I do not use the Prusa style and just inherited the code without ever changing it.  I look forward to seeing how that works with my printer.

Sebazzz, 2021-03-05:
Fair enough, I will remove the Prusa stuff completely. I will also clean up the layout of the checkboxes, so the checkboxes are at the left side of the label like in the rest of the Octoprint settings UI.

Sebazzz, 2021-03-13:
@tpmullan I submitted pull request #37 for this

tpmullan, 2021-03-21:
Thank you!  I will do some more testing and create a release.  For now it has been merged to master.

closed 2021-03-21 by tpmullan

Sebazzz, 2021-03-21:
Awesome, thank you for taking the time 😄

laz-, 2021-03-22:
I saw the prusa support in the README.md, came here to find that the option is now gone. It should probably get dropped from the docs too :)