tpmullan/OctoPrint-DetailedProgress #41: ETA display time seems to be ignoring system timezone

gdeconto, 2021-04-15:
I noticed that my ETA time shows as 7 hours more than what I manually compute the end time to be based on how many hours/min/etc are left to print.

It seems it may be ignoring the timezone I set; for example, my current (5:55PM) that will take about 1.5 hours more to print is showing 2:31AM finish time tomorrow instead of 7:30PM

AlDaveed, 2021-05-25:
I have the same issue, I can't figure what reference time is used for the  ETA so I can modify it.  My ETA is usually about  a day off.

dman97, 2021-07-09:
I have the same issue, is there a way to adjust this?

tnybrt, 2021-07-29:
Same thing but I'm unable to find a place to adjust it at

dman97, 2021-07-29:
> Same thing but I'm unable to find a place to adjust it at

I found the issue was that I did not have the time zone set in my Raspberry Pi.  Once I completed the localization settings the time was correct.

gdeconto, 2021-07-29:
Dman97, that was the first thing I checked.  The pi time is set correctly

ghod90, 2021-08-17:
Having the same issue

wjcarpenter, 2022-07-02:
I hope I'm not over-explaining this to folks who already understand this completely.

On Linux, the timezone is controlled by the TZ environment variable. Different users can have different preferences for that variable. If you run OctoPrint as a service or inside a Docker image, it could be that the TZ variable is not set correctly or not set at all. So, even if you see things in the correct timezone when you log onto your server, it can still be wrong for various services and processes that are not started from the command line.

(The sample docker-compose file given on the OctoPrint web site does the right thing for timezeon for the config editor but not for the octoprint process itself. That's how I went down this path.)

closed 2026-05-25 by tpmullan