tpmullan/OctoPrint-DetailedProgress #48: Percentage shown "12.96omplete" not "12% complete" or "12.69% complete"

tehniq3, 2023-04-01:
On ANET A8 LCD 2004 display shown with your plugin:
[image]
but other info are ok, "Good accuracy", "ETL 00h25m56s", "ETA 3/3 9.44PM"

without your plugin, last rowis empty
[image]

original, without Octoprint:
[image]

I try it a lot of version of text istead "{completion:.2f}% complete", adding spaces or change the text.. no good results...

tehniq3, 2023-04-02:
solved !!! solution is I found at https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/302 😁

you are right ! instead one % I put %%% 😎 I need to change in {completion:.2f}%%% done to see "16.4% done"
[image]
PS: many thank to you... nota some beer from my side ! 🤘🍺🍺🍺

tpmullan, 2023-04-02:
That is interesting. I guess the screen is thinking the % character is a command or something. If you want to create a PR and add info to the readme so others can easily fix this I can merge it.

closed 2023-04-02 by tpmullan